gamemanual0/gm0 #23: Wheel Guide

dansman805, 2020-08-12:
**What information should be added?**
An overall wheel guide should be added; this should be it's own page under drivetrains.

This would cover all types of wheels.

With this, you could make the [mecanum wheel miniguide](https://gm0.copperforge.cc/en/latest/docs/robot-design/drivetrains/holonomic.html#mecanum-wheels-miniguide) a subsection of it, and just link to it in the mecanum section. (Also make sure to cover the REV Mecanum wheels.)

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
It should be under https://gm0.copperforge.cc/en/latest/docs/robot-design/drivetrains/index.html.

shurik179, 2020-08-12:
I could try and write something, but not in the next two weeks...

dansman805, 2021-01-04:
To clarify, this issue is specifically a drivetrain wheel guide, though an intake specific wheel guide under the intake section is a good idea.

cadandcookies, 2021-04-14:
To clarify, is the intent of this to provide a similar comparison of wheel product offerings from various vendors for other common wheel types in FTC (eg, traction/grip, omni, compliant) as the mecanum mini-guide?

dansman805, 2021-04-14:
> To clarify, is the intent of this to provide a similar comparison of wheel product offerings from various vendors for other common wheel types in FTC (eg, traction/grip, omni, compliant) as the mecanum mini-guide?

Yeah, that is the intent.

dansman805, 2021-08-20:
Since most traction wheels perform relatively similarly, and are so simple, there's not much to say about them. There is not much none as far as I know about omni wheel performance. Therefore, documenting mecanums makes the most sense—and that is already done.

closed 2021-08-20 by dansman805