gamemanual0/gm0 #45: Add a contributing section

dansman805, 2020-09-26:
**What information should be added?**
Add a contributing guide so it's easier to contribute, look at the frc docs contributing guide. 

**Does it need it's own page? (Yes/No):** Yes (probably? Multiple) 

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
Probably in the appendix, or as it's own section.

dansman805, 2020-09-29:
See https://docs.wpilib.org/en/latest/docs/contributing/frc-docs/index.html

dansman805, 2021-01-30:
Make sure to mention making a venv in the env directory

NoahBres, 2021-01-30:
Add the venv in the gitignore after you do that

dansman805, 2021-01-30:
> Add the venv in the gitignore after you do that

It is? https://github.com/Coppersource/gm0/blob/1fde2a62bbf6b3616bd018f878a8483e3a48f135/.gitignore#L1

NoahBres, 2021-01-31:
oh i didn't realize you named it env
oops

dansman805, 2021-02-15:
Make sure to put the contributing guide at the end of gm0; put the contributors list under it and get rid of the meta folder as the other content is outdated.

dansman805, 2021-02-15:
See https://docs.google.com/document/d/1-YGhDATtziNO3iV1V4Kn8B5mEZ4NMYSrgLkZE4_MffA/edit, a style guide for gm0

closed 2021-04-04 by dansman805