gamemanual0/gm0 #46: Expand upon Control System Information

dansman805, 2020-09-29:
**What information should be added?**
Relevant information to programmers about how the control system works/performs; this is already covered partially at https://gm0.org/en/stable/docs/software/using-the-sdk.html#a-note-on-hardware-call-speed, but expanding upon it as it's own page would help make a centralized source of info about how the control system works.

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
It should be under the software or electrical section; https://gm0.org/en/stable/docs/power-and-electronics/index.html, https://gm0.org/en/stable/docs/software/index.html.

dansman805, 2020-09-29:
Probably put it in software and link to it from somewhere in electrical? (try putting it under both table of contents, see if that works idk)

dansman805, 2020-09-29:
Putting it under both tables of contents works; if it's referenced under electrical ToC as `../software/whatever` it will get linked to but will get skipped over in next from electrical, so that seems applicable here.

NoahBres, 2020-10-01:
Code samples for bulk reads and an explanation on the auto vs manual bulk read modes would be swell :D

NoahBres, 2020-10-04:
Explanation of the differences between the expansion hub and the control hub.

things to mention
- lack of ftdi for faster hardware calls
- lack of screen on chub. suggest alternatives like scrpy or just plugging in a monitor
- state that you still need an expansion hub for expanding inputs on the chub.

closed 2020-10-07 by dansman805