gamemanual0/gm0 #52: Simple go to point explanation/sample code

NoahBres, 2020-10-04:
**What information should be added?**
Simple go to point explanation utilizing PID on top of mecanum kinematics.

Things to touch on:
- explain that you should be using this with dead wheels. Go to point doesnt have any acceleration limiting so not really suitable for mecanum or you will experience high slip and it will throw off mecanum odometry. 
- basic go to point implementation. find angle between your current point (known via odo) and the target point. With a simple atan2 function. 
- Feed into trig based mecanum kinematics. Angle should be the angle you calculated. Magnitude/speed should be your result from the pid controller. Figure out this using math.hypot function. Then feed that distance into your PID controller.
- Explain how PID controller gains influence behavior. kP makes it move faster/more aggressively once bot nears the target but can overshoot. kD to control deceleration. etc. 

- Cool plus: Offer example code. gm0 example code repo? 

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
Some software page.

NoahBres, 2020-10-04:
Oops. Should be incorporated into #48

NoahBres, 2020-10-04:
Add trig mecanum kinematics under the existing mecanum kinematics if this page were to be added.

abidingabi, 2022-08-10:
I think the real target audience for go-to point sample code would be rookies who are largely better served with road runner; this could be a useful resource to have somewhere but probably not in gm0.

closed 2022-08-10 by abidingabi