gamemanual0/gm0 #53: Odometry page in hardware

NoahBres, 2020-10-04:
**What information should be added?**
Add an odometry hardware page. Explanation of what dead wheels are (and difference between odometry and dead wheels/tracking wheels). Go over common encoders like rev throughbore, e8t, srx mag, and the benefits. Maybe reference existing designs like open odo? Add pictures of people’s existing odo. Link to odometry code page.

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
Hardware

JIceberg, 2020-10-05:
Not too confident in my ability to write easy-to-understand mechanics of odometry hardware. Might require several reviews.

JIceberg, 2020-10-16:
@NoahBres What exactly do you mean by "hardware section"? Currently there is "Robot Design" and "Hardware Basics". I feel like it should go in robot design, right?

NoahBres, 2020-10-16:
That's a @dansman805 question to answer.

dansman805, 2020-10-16:
It'd go under "Robot Design"; however, you'd def want to mark it off as a *very* niche thing, given gm0's primary audience.

dansman805, 2020-10-21:
This got resolved with https://github.com/Coppersource/gm0/pull/50

closed 2020-10-21 by dansman805