gamemanual0/gm0 #54: Change mecanum kinematics to inverse kinematics

NoahBres, 2020-10-05:
**Where is there misleading or incorrect information?**
Pages referencing mecanum kinematics

**What's wrong?**
Calculating body velocity = forward kinematics
Calculating wheel velocities from desired body velocity = inverse kinematics

Clarification is probably good once kinematics start being mentioned for odometry. Odometry requires forward kinematics while what the kinematics refers to now is inverse.

dansman805, 2020-10-05:
I feel like we need to make an "advanced" section under software; I'm probably going to move FSMs and Odometry there, and then add the forward/inverse kinematics for some basic mechanisms (differential/mecanum drive, probably an arm?)

NoahBres, 2020-10-05:
That would be a good idea. Actually I just realized that the mecanum programming section doesn't even mention the word kinematics. Not sure why I thought it did. It just uses the phrase control equation. 
Nevertheless, kinematics are going to be mentioned with odo so I think clarification will be necessary. 
But yea. A list of common kinematics would be coolm

PR #263 (2022-08-12): Add Kinematics Page
Adds concept page to move tutorials that aren't actually tutorials.

Closes #54
closed 2022-08-12 by abidingabi