gamemanual0/gm0 #57: Expand upon Information about Java

dansman805, 2020-10-11:
**What information should be added?**
This page is lacking depth. Revamp it. Make sure to cover: control flow, OOP, variables, methods, and some basic data structures (in no particular order).

**Does it need it's own page? (Yes/No):** No

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
https://gm0.org/en/latest/docs/software/fundamental-concepts.html

Aman-Anas, 2020-10-11:
input

Aman-Anas, 2020-10-11:
Maybe cover more detail about getting started with the FTC development environment, whether it is Android Studio, OnBot Blocks, or OnBot Java. The wiki page on the FTC SDK isn't too detailed, plus doesn't cover a lot of typical issues and errors which may occur (whether it's from Gradle, or the Android SDK license, or so on).

dansman805, 2020-10-11:
That'd probably go under using the SDK instead of programming fundamentals, but that's definitely fair

NoahBres, 2020-10-12:
+1 on getting started. The resources for that are pretty bad. There are some good videos but none in concise text form.

dansman805, 2020-10-12:
Looking back, I feel like a "Android Studio" or "Getting Started with Android Studio" or something page is deserved for this topic specifically, as opposed to generic programming concepts; see https://github.com/Coppersource/gm0/issues/59.

dansman805, 2021-01-05:
See https://docs.google.com/document/d/1KCE6ukW66f0SYM-qc3ICf5FqCL4gTQc1TlmuQjf_UaI/edit#

abidingabi, 2022-07-26:
A java guide is out of scope for gm0

closed 2022-07-26 by abidingabi