gamemanual0/gm0 #59: Android Studio Basics

dansman805, 2020-10-12:
**What information should be added?**
Add info on how to use Android Studio, adb, gradle, common errors, etc.

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
Under "fundamental concepts of programming" or "using the sdk" in software https://gm0.org/en/stable/docs/software/index.html.

The latter *probably* makes more sense, as some people using java will be using OnBot, but if anybody has input i'd be glad to listen them.

NoahBres, 2020-10-12:
Possible outline for the page:
- Downloading/Installing AS
- Opening/importing the sdk project
- basic overview of the layout (screenshot of Android studio with a project open, highlight the code, the folder on the side, logcat button on the button, terminal button on the bottom, point out run button, clean button, etc)
- Gradle troubleshooting
  - gradle clean and rebuild button
  - invalidate cache and restart

dansman805, 2020-10-12:
Also have info on adb, see https://github.com/Coppersource/gm0/issues/48

JIceberg, 2020-10-21:
> srcrpy

Is that actually commonly used by teams? I don't know how many people use it.

dansman805, 2020-10-21:
It's not commonly used by teams, but it's really useful regardless so I think at least mentioning it and having a link is warranted

JIceberg, 2020-10-21:
> It's not commonly used by teams, but it's really useful regardless so I think at least mentioning it and having a link is warranted

Sounds good. I'll make sure to include it. Look out for the draft PR coming soon 😉

dansman805, 2020-10-22:
Also have info on how you need to change to photo sharing or file sharing and not "charging only" on the phone to use adb

shaunsingh, 2020-12-05:
Ill try working on this tomorrow (shaurya)

dansman805, 2020-12-31:
Closed by https://github.com/Coppersource/gm0/pull/69

closed 2020-12-31 by dansman805