gamemanual0/gm0 #64: Example code for runmodes

NoahBres, 2020-10-22:
**What information should be added?**
Write example code for runmodes. I specifically want to write samples for the setVelocity() arguments and lifting the max rpm fraction. 
Also sample code for RTP and it's correct order, just to be safe.

**Does it need it's own page? (Yes/No):** No

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
N/A

NoahBres, 2020-10-24:
Mention how setPower() actually controls the max power in RTP. Default is zero so remember to set it.

NoahBres, 2020-10-25:
Disclaimer that you probably shouldn't be using RUE for telex

NoahBres, 2020-11-26:
Mention that setVelocity turns on velocity control regardless of whether or not you set RUE

dansman805, 2021-11-28:
https://docs.revrobotics.com/rev-control-system/programming/using-encoder-feedback#choosing-a-motor-mode seems to cover this pretty well, linking to it is probably acceptable

abidingabi, 2022-08-09:
it is now https://docs.revrobotics.com/duo-control/programming/using-encoder-feedback#choosing-a-motor-mode

PR #270 (2022-08-12): Split up Using the FTC SDK page
Closes #64 
Closes #256 
Closes #86
closed 2022-08-12 by abidingabi