gamemanual0/gm0 #65: Add info for Vex 393 motor under the CR Servo section

NoahBres, 2020-10-22:
**What information should be added?**
Add a quick note that you treat vex 393 motors like cr servos and that you need to connect to an SPM. Digression but also maybe mention spm under the servo section?

**Does it need it's own page? (Yes/No):** No

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
"Examples of using common hardware components" -> CR Servo section

dansman805, 2020-10-22:
see https://gm0.org/en/stable/docs/hardware-basics/servo-guide/specialty-servos.html#vex-393-edr

NoahBres, 2020-10-22:
Tiny addendum: 

"The 393 motor acts as a more powerful and faster continuous rotation servo and is treated as a continuous rotation servo from the SDK when programming"

closed 2020-10-24 by dansman805