gamemanual0/gm0 #70: Blocks samples

NoahBres, 2020-10-25:
**What information should be added?**
Basic blocks samples. Mecanum drive blocks snippet especially. Would be impossible for a normal blocks team to figure out without knowing Java. And I've seen multiple blocks teams use mecanum drive. They just don't know how to program it. Also maybe just like simple tank drive since that's the most common. Pictures will be important imo to be digestible for a blocks audience. 

Also maybe recreate the official ftc blocks tutorial in gm0? I think gm0 could do well and expand upon it a bit. 
https://www.firstinspires.org/sites/default/files/uploads/resource_library/ftc/blocks-programming-autonomous.pdf
It's not actually that bad. (moreso just a problem with the blocks api itself being too complex imo)
My primary issue is that they show you how to set target position with the RTP but don't actually tell you what that target position implies. Idk if circumference * ticks per rev is the most understandable thing for a blocks audience but I do think it should be explained. 

General blocks samples would expand gm0 to the blocks audience too. 

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be in?**
¯\_(ツ)_/¯

dansman805, 2020-10-26:
I'd use sphinx tabs code tabs for that https://github.com/executablebooks/sphinx-tabs

NoahBres, 2020-10-27:
The sphinx tabs don't seem ideal for pictures? And the block samples would be pictures no?

colloquist, 2021-11-17:
See also https://github.com/executablebooks/sphinx-design for tabs

abidingabi, 2022-07-26:
The mecanum page has blocks examples.

closed 2022-07-26 by abidingabi