gamemanual0/gm0 #72: gm0 software section on button inputs

NoahBres, 2020-10-27:
**What information should be added?**
Specifically information on how to to read single button presses rather than just a constant boolean that returns true. Like an "isPressed" thing. I feel like this confuses beginners. 

**Does it need it's own page? (Yes/No):** 🤷 

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
🤷

JIceberg, 2020-11-01:
It definitely is something that confuses people. Maybe just make a quick reference guide for input-handling that also references the reversed values on the sticks. Not sure about how that would be formatted/organized.

dansman805, 2020-11-01:
I'd change Using the SDK into a section, put the current contents of it under a "Method Summary" or something, then write more detailed stuff under that (so have gamepad input be it's own page under Using the SDK).

dansman805, 2020-11-01:
By making a "Using the SDK" section, you could move "Control System Internals" under it too, as the main point of it is largely to explain bulk reads.

JIceberg, 2020-11-01:
> I'd change Using the SDK into a section, and put the current contents of it under a "Method Summary" or something, and then write more detailed stuff under that section (so have gamepad input be it's own page under Using the SDK).

#74 - I agree

NoahBres, 2021-02-05:
Mention "Failling edge detector"

dansman805, 2022-02-08:
Could should be straightforward, around like
```java
public void runOpMode() {
  private Gamepad previousGamepad1 = new Gamepad();
  private boolean toggleIntake = false;  
  
  while (opModeIsActive())
    if (gamepad1.a && !previousGamepad1.a) {
      // rising edge
    }

    if (!gamepad1.b && previousGamepad1.b) {
      // falling edge if you need that for some reason
    }

    if (gamepad1.x && !previousGamepad1.x) {
      toggleIntake = !toggleIntake; // rising edge detector that swaps toggle
    }

    // the actual toggle code
    if (toggleIntake) {
      // whatever
    }
    else {
      // the other thing
    }

    previousGamepad1.copy(gamepad1);
  }
}```

dansman805, 2022-02-08:
A dedicated gamepad page should also probably include gamepad diagrams (see https://github.com/gamemanual0/gm0/issues/114)

PR #245 (2022-07-25): Add gamepad input page
Closes #72 and #114.
closed 2022-07-25 by abidingabi