gamemanual0/gm0 #74: Expand Upon Using the SDK

JIceberg, 2020-11-01:
**What information should be added?**
The current format fits closer to that of a pdf--which was the previous intent. There are still several concepts that need to be addressed as well as functionalities (like gamepad inputs).

**Does it need it's own page? (Yes/No):** Split it into subpages.

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
See above.

dansman805, 2020-11-01:
Ideally, we'd have separate pages for:

* OpMode/LinearOpMode
* HardwareMap
* Motors
* Servos? idk if this is as necessary, maybe could share a spot with motors under an "actuators" page
* Gamepads
* Sensors (the following should be mentioned)
  * IMU
  * Digital
  * Analog
  * Maybe like the rev TOF and color sensor 
* Bulk reads (as part of control system internals)
* Method Summary (roughly in the style of the current page)? tbh i don't think this makes sense, but 🤷

Eeshwar-Krishnan, 2022-08-09:
I think so far we have most of these, I mentioned in another issue that the SDK has example things for all of these so maybe a page about the SDK examples and grouping them with brief explanations of what they are/how to access could close the rest

closed 2022-08-10 by abidingabi