gamemanual0/gm0 #78: Add bulk read clarification

NoahBres, 2020-11-09:
**What information should be added?**
Add details on what a bulk read is.
- How it reads analog/DIO/etc. in a single read
- Each rev hub has a single bulk read
- also maybe add NPE's detail about the slight delay when the rev hubs have to repeat commands over the 485
- how i2c cannot be part of bulk reads

**Does it need it's own page? (Yes/No):** No

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
Software > Using the FTC SDK > A Note on Hardware Call Speed

dansman805, 2020-11-09:
Does this not cover it? https://gm0.org/en/latest/docs/software/control-system-internals.html#bulk-reads

NoahBres, 2020-11-29:
oops yea. originally it didn’t show up in the search when i was trying to find it.

closed 2020-11-29 by NoahBres