gamemanual0/gm0 #86: External encoders and programming them

NoahBres, 2020-11-29:
**What information should be added?**
Explain how external encoders plug into a motor encoder port and should be treated like a motor in the sdk.

**Does it need it's own page? (Yes/No):** No

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
Programming the sdk section?

dansman805, 2020-12-04:
IMO, make a "sensor programming " page or something, cover encoders in general (CPR vs PPR, etc), have a subsection for motor encoders (go over run modes), and have a subsection for in external encoders (cover the stuff in parent, velocity overflow, and the encoder wrapper).

JIceberg, 2020-12-08:
I have a sample for this in FTCLib. Should I make something similar using the SDK or do you want to use that as a sample? Might be useful to have a sample (code samples are useful of course!)

PR #270 (2022-08-12): Split up Using the FTC SDK page
Closes #64 
Closes #256 
Closes #86
closed 2022-08-12 by abidingabi