gamemanual0/gm0 #87: Basic time based auto sample

NoahBres, 2020-11-30:
**What information should be added?**
Basic sample of time based auto. `while(timer.seconds() < setTime) { move() }`
Explain why this isn't ideal but it does work fine as a first step. Also the fact that the while loops are blocking.

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
🤷‍♂️

Eeshwar-Krishnan, 2022-08-09:
I kinda disagree with the addition of this page, given the sdk has examples for many things like this. Maybe instead a page linking the SDK examples and a brief explanation of them would be good, however that would have to be regularly maintained given the example are sometimes changed/renamed

Eeshwar-Krishnan, 2022-08-14:
This is out of scope for gm0, sdk examples are linked to

closed 2022-08-14 by Eeshwar-Krishnan