gamemanual0/gm0 #89: Add vision page

dansman805, 2020-12-05:
**What information should be added?**
Talk about Vuforia, Tensorflow, (Easy)OpenCV; mention scrcpy as an option to look at camera output.

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
Make an advanced section under software, then put it under that.

shaunsingh, 2020-12-05:
Ill try working on this tomorrow (shaurya)

shaunsingh, 2020-12-06:
https://docs.google.com/document/d/1w6yG1Ph9jYKsGFOdHocUbrGFC1ooq7zbf-JrEdTkf6M/edit?usp=sharing

draft

dansman805, 2020-12-07:
> Note: Tensorflow cannot run on the older ZTE speed phones, and need a phone with android 6.0 (marshmallow) or higher

All legal android phones can run Tensorflow, and the minimum version for every legal phone is at least marshmallow so this note is redundant.

shaunsingh, 2020-12-07:
I modified it to remove the redundant lines. I also gave you edit permissions if you think anything else needs editing.

NoahBres, 2021-01-30:
Basic EOCV pipeline sample? Basic pixel sampling perhaps?

shaunsingh, 2021-02-07:
eocv section updated with link to examples

shaunsingh, 2021-02-07:
https://github.com/shaunsingh/gm0/blob/master/source/docs/software/vision.rst

Added file is here. Could someone please test before I PR as I am having issues with my make install

shaunsingh, 2021-02-10:
https://github.com/Coppersource/gm0/pull/145

dansman805, 2021-02-12:
Closed by https://github.com/Coppersource/gm0/pull/145

closed 2021-02-12 by dansman805