gamemanual0/gm0 #101: Telemetry feature documentation

NoahBres, 2021-01-04:
**What information should be added?**
Document telemetry features
- Supported telemetry tags w/examples
- Monospace telemetry, etc.
- Changing transmission interval

**Does it need it's own page? (Yes/No):** No(?)

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
Under "Using the SDK" maybe?

NoahBres, 2021-02-04:
`addLine` vs `addData`

NoahBres, 2021-02-24:
https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/text/Html.java#791

Supported tags:

- `<br>` (doesnt actually do anything)
- `<p>`
- `<ul>`
- `<li>`
- `<div>`
- `<span>`
- `<strong>`
  - `<b>` (does the same thing)
- `<em>`
  - `<cite>`
  - `<dfn>`
  - `<i>`
  - (all of these just italicize)
- `<big>`
- `<small>`
- `<blockquote>`
  - seems to just be a wrapping element like span
- `<font>`
  - also seems to just be a wrapping element like span?
- `<tt>`
  - monospace wrapper
- `<a>`
  - it doesnt look like it uses an actual link in the source? so it's just a wrapper element like span?
- `<u>`
  - underline
- `<del>`
  - `<s>`
  - `<strike>`
  - all strikethrough elements
- `<sup>`
  - superscript
- `<sub>`
  - subscript
- `<img>`
  - img doesnt actually do anything. the src attribute seems to exist only to be handled on the android app side and because we dont have access to the DS we can't do anything with it
- `<h1-h6>` tags

None of these have been tested

NoahBres, 2021-02-24:
`<font size="..." color="..." face="...">`

https://www.techonthenet.com/html/elements/font_tag.php

size = relative size? like `size="+1"`
needs to be tested

color = "string" hex or named colors
source looks like it supports hex and then only a few named colors.
https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/text/Html.java#694
needs to be tested

face = "string"
needs to be tested. "monospace" works as a value for sure.
I believe it should support generic-family and family-family from here: https://www.techonthenet.com/css/properties/font_family.php
it seems to just pass the string value into an android font object https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/text/Html.java#1153
so im betting it's fairly flexible

NoahBres, 2021-02-24:
`<div>` seems to support an align "start", "center", or "end"
not entirely sure how this works. would depend on the width that the div tag takes up. if it fills an entire line these would work as expected. needs to be tested.

NoahBres, 2021-02-24:
https://github.com/OpenFTC/Extracted-RC/blob/6720cf8b4296c90b6ea4638752c2df4a52b043b9/RobotCore/src/main/java/org/firstinspires/ftc/robotcore/external/Telemetry.java#L542

`DisplayFormat` options `CLASSIC`, `MONOSPACE`, `HTML`

NoahBres, 2021-02-24:
https://github.com/OpenFTC/Extracted-RC/blob/6720cf8b4296c90b6ea4638752c2df4a52b043b9/RobotCore/src/main/java/org/firstinspires/ftc/robotcore/external/Telemetry.java#L79

This sample is interesting. Never seen it used before.

NoahBres, 2021-02-24:
https://github.com/OpenFTC/Extracted-RC/blob/6720cf8b4296c90b6ea4638752c2df4a52b043b9/RobotCore/src/main/java/org/firstinspires/ftc/robotcore/external/Telemetry.java#L104

> Actual transmission to the driver station is throttled to avoid use of excessive bandwidth.
> By default, transmission will occur at most every 250ms. This interval can be controlled with
 > {@link #setMsTransmissionInterval(int) setMsTransmissionInterval()}. Any {@link #update()}s which
> occur more frequently will not be transmitted if superseded by a subsequent {@link #update()}
> before the transmission interval expires.</p>

NoahBres, 2021-02-24:
https://github.com/OpenFTC/Extracted-RC/blob/6720cf8b4296c90b6ea4638752c2df4a52b043b9/RobotCore/src/main/java/org/firstinspires/ftc/robotcore/external/Telemetry.java#L137

[image]

NoahBres, 2021-02-24:
`clear()` vs `clearAll()`

The telemetry api is lowkey not very straightforward because of the fancy stuff they added

NoahBres, 2021-02-24:
`Telemetry#speak()`

NoahBres, 2021-02-24:
The telemetry api is sooo over-engineered 

[image]

NoahBres, 2021-02-24:
Items can have captions

[image]

dansman805, 2021-04-21:
Realistically, when this is worked on, we should definitely write something containing just `addData`, *maybe* `addLine`, transmission interval, and monospace/html. Telemetry is very complicated, but a significant majority of the value of it can be obtained from those few things.

NoahAndrews, 2021-04-21:
I would add `speak()` to that list, as it adds significant value that can't be approximated using other APIs.

PR #269 (2022-08-12): Add Telemetry Page
Closes #101
closed 2022-08-12 by abidingabi