gamemanual0/gm0 #102: Mentor Guide

quinn-mcginley, 2021-01-04:
**What information should be added?**
Information regarding how to approach mentoring teams, properly support team members, and foster a productive learning environment. Topics covered may include how to motivate students, how to settle disputes, how to effectively teach members new skills, how to seek out mentors in your community, and how to ensure longevity of teams. [Doc of why this is important](https://docs.google.com/document/d/1o1zWYinDUxiXQ8e1gcm_FniMnYrpGjIg8rpp5VK37As/edit?usp=sharing) (Not OC)

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
It could go in the ["Getting Started in FTC" page](https://gm0.org/en/stable/docs/getting-started-in-ftc/index.html#), although it might work better in a section of its own, especially if new pages are added related to strategy, outreach, management, etc.

dansman805, 2021-01-05:
See FTC's official mentor guide: https://www.firstinspires.org/sites/default/files/uploads/resource_library/ftc/mentor-manual.pdf

And one from a while ago (new one seems to have some content missing like the "I Do, You Watch" to "You Do, I Watch progression): https://web.archive.org/web/20120829092117/http://www.usfirst.org/sites/default/files/uploadedFiles/Robotics_Programs/FTC/FTC_Documents/FTCMentorGuide_2012.pdf

abidingabi, 2022-07-26:
This is out of scope for gm0, other resources exist on the topic

closed 2022-07-26 by abidingabi