gamemanual0/gm0 #103: NPE Exception Debugging

NoahBres, 2021-01-05:
**What information should be added?**
NPE exceptions are sooo common and most beginner programmers often dont know how to handle them. Explain what null is? Explain why it happens. Etc. Hardwaremap being null because you're declaring it inside the class fields rather than in runOpMode. etc.

Needs a better outline tbh. 

Explain how wireless adb will literally give you the exact line that the error occurs in.

**Does it need it's own page? (Yes/No):** idk

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
No clue 🤷‍♂️

JIceberg, 2021-01-13:
Would probably go in the Java page under a section called debugging. I think the Java page needs to be restructured anyways as sub-pages that go over the basics in more detail anyways (#57).

NoahBres, 2021-01-13:
See https://docs.google.com/document/d/1KCE6ukW66f0SYM-qc3ICf5FqCL4gTQc1TlmuQjf_UaI/edit?usp=sharing for Java page restructuring

JIceberg, 2021-01-13:
> See https://docs.google.com/document/d/1KCE6ukW66f0SYM-qc3ICf5FqCL4gTQc1TlmuQjf_UaI/edit?usp=sharing for Java page restructuring

Porting or add as a useful resource/link on the Java page?

NoahBres, 2021-01-13:
Afaik, plans are to incorporate that into gm0. Would talk to @dansman805 about comprehensive details on that.

Eeshwar-Krishnan, 2022-06-07:
Closed by #216

closed 2022-06-07 by Eeshwar-Krishnan