gamemanual0/gm0 #104: goBILDA product insights like thing for common FTC mechanisms

NoahBres, 2021-01-05:
**What information should be added?**
goBILDA product insights like thing for common FTC mechanisms. CAD pictures of things like lifts, intakes, sprung intakes? flywheels, linked motors?, etc. Just basic mechanisms with assembled CAD, blown up CAD pictures, etc.

**Does it need it's own page? (Yes/No):** Probably

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
🤷‍♂️

NoahBres, 2021-01-05:
~~have gobilda sponsor this page and only use gobilda COTS for all of the mechanisms~~

dansman805, 2021-01-05:
> ~have gobilda sponsor this page and only use gobilda COTS for all of the mechanisms~

To only use goBILDA COTS we just need Imants to do the CAD :stuck_out_tongue:

dansman805, 2021-01-05:
A couple things, I'm not really sure what such a page would be called, but we'd also want to add a disclaimer saying "don't copy these directly, they're just for inspiration, etc."

NoahBres, 2021-01-05:
> To only use goBILDA COTS we just need Imants to do the CAD 😛

Outreach++

> A couple things, I'm not really sure what such a page would be called, but we'd also want to add a disclaimer saying "don't copy these directly, they're just for inspiration, etc."

"Design inspiration"? I think it merits a cooler name tho. Idk. 

> but we'd also want to add a disclaimer saying "don't copy these directly, they're just for inspiration, etc."

True. But personally don't think that's a huge issue. It's more akin to sample code. Like you can never really directly copy anything anyways since integration with your own components is a challenge. But still a good disclaimer.

PR #252 (2022-08-04): Add common mechanism models
closes #104
closed 2022-08-04 by abidingabi