gamemanual0/gm0 #106: Add ACME Guide

NoahBres, 2021-01-05:
**What information should be added?**
https://cdn.discordapp.com/attachments/551473381002706964/585339484380594196/AR-040619-0528.pdf
A reminder to include the one ACME guide since it still hasn't been added :P (not that I'm actually helping ameliorate that issue but I also have no idea where it would go)

**Does it need it's own page? (Yes/No):** No clue

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
No idea

dansman805, 2021-01-05:
Probably under useful resources?

dansman805, 2021-01-29:
Make sure to throw it on google drive or something, linking to a discord attachment seems not ideal

Eeshwar-Krishnan, 2022-08-12:
Honestly the only really useful thing there is to use !isStopRequested() instead of opModeIsActive()

Eeshwar-Krishnan, 2022-08-12:
Closed as the note about opModeIsActive is already in there

closed 2022-08-12 by Eeshwar-Krishnan