gamemanual0/gm0 #111: Mention monkey command for launching app

NoahBres, 2021-01-08:
**What information should be added?**
Android Studio seems to not automatically launch the app a lot of the time so when you press play and upload the app, the app doesn't actually start. So you have to go to your phone and manually open it up. The workaround I found for this was running this command through ADB:

```bash
adb shell monkey -p com.qualcomm.ftcrobotcontroller 1
```

This way, you don't need to touch your phone.
The conventional way of start an app `adb shell start -n com.package.name/com.package.name/ActivityName` does not seem to work for me.

**Does it need it's own page? (Yes/No):** No

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
Idk wherever the ADB stuff is.

NoahAndrews, 2021-01-09:
Won't that monkey command send a random input to the app? That feels like a bad way to launch it. The standard way should work fine, I'm surprised you said it didn't.

NoahBres, 2021-01-10:
Well, to add onto the many things I have been wrong about in the past few days, adb shell start does in fact work. I am quite sure it did not work at one point? I documented which commands I tried in the Discord and asked about when originally trying it out last year. I guess I missed something. Didn't have anyone to correct me :P

Specifying 1 as the last argument for that command will default to the launch app intent so it should be fine. Although probably not ideal :P

Eeshwar-Krishnan, 2022-08-09:
The rev hardware client seems to have an option for starting the app, maybe that would be more useful instead of adb start

Eeshwar-Krishnan, 2022-08-14:
Closed by https://github.com/gamemanual0/gm0/pull/276

closed 2022-08-14 by Eeshwar-Krishnan