gamemanual0/gm0 #113: Updating your sdk via git

NoahBres, 2021-01-10:
**What information should be added?**
Updating your project to the latest SDK using git upstream
- Set git upstream
- Git fetch/merge or git pull
- Light explanation on what merge conflicts are

**Does it need it's own page? (Yes/No):** Probably not

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
/software/using-android-studio.html#version-control might work depending on the depth

Eeshwar-Krishnan, 2022-08-13:
Due to how quickly this becomes a very advanced topic (having to deal with merge conflicts and such, it quickly becomes a whole how to use git tutorial) this is probably out of scope for gm0

closed 2022-08-13 by Eeshwar-Krishnan

abidingabi, 2022-08-13:
See https://github.com/gamemanual0/gm0/pull/170#issuecomment-1214243403