gamemanual0/gm0 #114: Gamepad Diagrams

NoahBres, 2021-01-13:
**What information should be added?**
Gamepad diagrams
Ideally for all 3 of the supported gamepad (etpack is just a dualshock clone so can be excluded, plus it's ugly): Xbox 360, Dualshock 4, and Logitech F310

A list of the button aliases would be nice.

**Does it need it's own page? (Yes/No):** Idk

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
🤷

NoahBres, 2021-01-13:
Relevant for aliases: https://github.com/OpenFTC/Extracted-RC/blob/6720cf8b4296c90b6ea4638752c2df4a52b043b9/RobotCore/src/main/java/com/qualcomm/robotcore/hardware/Gamepad.java#L767

dansman805, 2021-01-18:
Include positive/negative directions on joysticks (up is negative, left is negative, down is positive, right is positive)

ajtrowell, 2021-02-13:
[image]
[image]

shaunsingh, 2021-02-13:
[image]

[image]

They don't look as good as the top two images but most diagrams are of ps3 dualshocks

dansman805, 2022-02-08:
https://commons.wikimedia.org/wiki/File:360_controller.svg

closed 2022-07-25 by abidingabi