gamemanual0/gm0 #128: Styling Tweaks

NoahBres, 2021-01-30:
Various styling tweaks I would like to make

- Increase heading font weights
  - h1: 500
  - h2: 400
- List styling
  - Padding left for lists so they don't align with headings
  - Margin below each image in a list 
  -  [image]

- Note block tweaks
  - idk why but this looks off to me
  - [image]

NoahBres, 2021-01-30:
[image]

These sections all blend together. Increase division somehow. Increasing font weight for heading may be enough.

NoahBres, 2021-01-30:
Improve inline code. There's like no contrast between the background of the inline code and the white background. Maybe more padding

[image]

NoahBres, 2021-01-30:
[image]

Linked terms mix different underlines.

dansman805, 2021-02-16:
> Linked terms mix different underlines.
Most of those are glossary items and therefore show something on hover; normal links do not and therefore do not get an underline. Now, whether that's a good thing, :shrug:

closed 2022-07-26 by abidingabi