gamemanual0/gm0 #150: Thoughts on mentioning common wheels on the dead wheel page?

NoahBres, 2021-02-12:
**What information should be added?**
Would it be helpful to list common wheels for dead wheel 

**Does it need it's own page? (Yes/No):** No

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
docs/robot-design/dead-wheels.html

dansman805, 2021-02-12:
Probably a good idea to list a couple, should also probably talk about single vs double omni for odo

NoahBres, 2021-02-12:
> Probably a good idea to list a couple with pros/cons, should also probably talk about single vs double omni for odo

What should be mentioned for single vs double?

dansman805, 2021-02-12:
> What should be mentioned for single vs double?

Why single omni odometry doesn't work well, and how double omni alleviates those issues

dansman805, 2021-02-12:
> What should be mentioned for single vs double?

Why single omni doesn't work well, and how double omni alleviates those issues

Eeshwar-Krishnan, 2022-06-07:
Honestly, realistically, just about every odometry pod uses the same rotacaster 35mm omni wheels, so I don't know if this would ultimately be useful

closed 2022-06-07 by Eeshwar-Krishnan