gamemanual0/gm0 #151: Mention why "Stuck in stop" occurs

NoahBres, 2021-02-12:
**What information should be added?**
[image]

**Does it need it's own page? (Yes/No):** No

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
Idk. One of the software pages about the SDK

dansman805, 2021-02-12:
Maybe a "common issues" or something page, that would also work with https://github.com/Coppersource/gm0/issues/103

Eeshwar-Krishnan, 2022-06-07:
I believe that https://gm0.org/en/latest/docs/software/common-issues.html#stuck-in-start-loop-stop addresses this

closed 2022-06-07 by Eeshwar-Krishnan