gamemanual0/gm0 #190: Add page on how to use tools accessible to FTC teams

jackcrane, 2021-07-27:
**What information should be added?**
The FTC api and the TOA api are both extremely powerful ways to collect data. Unfortunately the official FIRST events api documentation is old and not specific and the TOA api documentation is new but poorly documented. It would be nice to have an explanation on how these tools are used with a clean and useful documentation section

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**

https://gm0.org/en/latest/docs/software/using-the-apis.html

dansman805, 2021-07-27:
Honestly, in my opinion the FTC events API is documented pretty well as it is—I had issues with getting authenticated but that was me missing a key part of the process that was plainly written. Anything written here would likely be less comprehensive than the actual documentation—most of the actionable stuff like which endpoint has what data is documented pretty well. That would leave more user friendly documentation to mostly be how to authorize when connecting to the API, which is covered in the [authorization section](https://ftc-events.firstinspires.org/api-docs/index.html#section/Authorization) and how to make a web request and perhaps parsing JSON in whatever language—which while definitely useful information, isn't novel or FTC specific enough to be included here in my opinion. 

Using the API isn't even *that* useful for e.g. scouting in FTC—given most events don't have live data and historical data from however many months ago isn't really pertinent to current robot performance.

While it'd be cool to have some more user-friendly documentation on it, to me this seems like something that's realistically pretty niche, would require quite a bit of work to write, and would require nontrivial upkeep when the API get updated. 

Also, as a side note, I'd definitely not put it under "using-the-apis" because that is very easily confused with the FTC SDK itself—maybe something like "using-ftc-events-api" fits better.

jackcrane, 2021-07-27:
That is true. gm0 may not be the best place for this either. 

closing #190

closed 2021-07-27 by jackcrane