gamemanual0/gm0 #230: VEX 393 Motor

AlecHub, 2022-06-01:
**Firstly**, many thanks to all the contributors of gm0.

The [VEX 393 EDR](https://gm0.org/en/latest/docs/power-and-electronics/servo-guide/specialty-servos.html?highlight=393#vex-393-edr) section says "_The VEX 393 EDR is technically a motor, but is legally counted as a servo in Game Manual 1 rules. **Why this is, we do not know**_, ..." 

The [Glossary](https://gm0.org/en/latest/docs/appendix/glossary.html#term-Servo) says "**A servo is a small DC motor ...**"  

Since small DC motors are counted as servos, it makes sense that the VEX 393 motor is counted as a servo".  Besides, the VEX 393 motor kinda looks like a servo.

I believe the distinction between a small DC motor and a regular DC motor is the power source:  If the motor is powered at 6V or less, then it is a small DC motor (i.e. a "Servo").  Otherwise it is a regular DC motor (i.e. a "DC Motor").

AlecHub, 2022-06-02:
I suggest rewriting the entire [VEX 393 EDR](https://gm0.org/en/latest/docs/power-and-electronics/servo-guide/specialty-servos.html?highlight=393#vex-393-edr) section.  Very little of the section is accurate.

Eeshwar-Krishnan, 2022-06-06:
While I agree that there is some sense as to why its considered a Servo, I'm confused as to why "very little of the section is accurate." I don't see anything particularly egregious, faster cr servos may be available now but as to pure power the 393 is still one of the most powerful CR servos that can legally be used

Eeshwar-Krishnan, 2022-06-07:
Information on the 393 page was reviewed and deemed accurate

abidingabi, 2022-06-07:
The VEX 393 (2-wire) is distinct from normal Servo as it does not take 3-wire PWM input and requires an external motor controller, however for FTC purposes it (in conjunction with an MC29) is legal and counts as a Servo. 

*Why* something is legal is not in the game manuals—only whether it is or is not legal.  *Why* the VEX 393 is legal as a servo probably has something to do with making it easier/less expensive for teams to switch from VRC to FTC, but any comment about that would be speculation. The point about it being legal being an exception is because it is the only thing in FTC that can be used as a Servo that requires an external 3-wire PWM-in motor controller.

The SPM description has the wrong voltage numbers (should be 4.8 and 6), but other than that, seems to be correct. The MC29 description is phrased a little bit weirdly, and would likely make more sense if phrased the other way (it takes a 3-wire PWM signal as input and provides a 2 wire motor output (which is also probably PWM, but since I don't have one and an oscilloscope in front of me I cannot verify that, and regardless it's a different PWM)). 

As far as I am aware, no FTC build systems (unless you count VRC parts, which while technically an option is not really a serious contender) have a native way to mount a 1/4" shaft.

abidingabi, 2022-06-07:
tl;dr "acts as a more powerful and faster continuous servo" should remove faster, SPM voltage numbers should be fixed, MC29 definition should clarify input vs output

closed 2022-06-07 by abidingabi

AlecHub, 2022-06-08:
One of the key things that sets FTC apart from other robotics leagues is that FTC allows teams to use virtually any construction material.  **There is no such thing as a "FTC Build System"**.

For teams that want to use VEX structural components for construction material, the utility of VEX structural components is greatly enhanced by the VEX 393 Motor.  Since the VEX 393 Motor is compatible with the FTC control system, it follows that the VEX 393 Motor is allowed.

You may not see any value in VEX structural components and mechanisms, **but FIRST does**.  There is plenty of educational value in using VEX structural components and mechanisms and the VEX 393 Motor greatly enhances the value. 

FIRST isn't trying to make it easy to switch from VRC to FTC. FIRST is just trying to be true to its calling to allow any robot construction material.

So how do you classify the VEX 393 Motor.  Is it a "**DC Motor**" type of motor or is it a "**Servo**" type of motor?

colloquist, 2022-06-08:
> There is no such thing as a "FTC Build System".
While this may be true in theory, in practice there's a reason Game Manual 0 lists tetrix, rev, actobotics, and goBILDA build systems specifically—the latter two being officially supported through the FIRST Storefront, the latter two being used by a majority of competitive teams at various points during gm0's activity. Yes, you can use anything to build FTC bots, and you can build competent ones out of primarily VRC parts. However, this has never really been common within competitive teams (excluding FVC), and I for one cannot name a team competitive on the world scale doing so since The Q is Silqent in Velocity Vortex. 

Yes, building FTC robots out of VRC structure can be done, and has about as much educational value as doing so out of any other build system—but it is not common for competitive teams, or in my experience common in general. Game Manual 0 does not have the contributor bandwidth and it does not make logistical sense to try to be all things to all people. A similar issue is directly brought up in gm0's mission statement:
> When perusing this guide, it is important to keep in mind the authors’ perspective. Many of the teams who contributed in the writing of this guide are veteran championship level teams in the upper echelon of FTC. This means that most of our recommendations are almost solely from the competitive advantage standpoint. For example, teams can learn just as much from using a Tetrix kit compared to an Actobotics kit. Just because there is a competitive advantage to one does not translate into a learning disadvantage for the other. The guide has a plethora of useful information and knowledge (such as FTC and basic engineering principles) for all teams, including teams which are primarily trying to learn rather than compete. However, do take our recommendations with a grain of salt. We want to help all teams in FTC, but as our experience has been on the competitive side of the fence, certain parts of the guide may not be very applicable to some teams.
If you're doing FTC primarily for education, that's perfectly fine, and arguably preferable to some extent—however, that's not what Game Manual 0 is aimed at. So while yes, there is not a specific “FTC Build System”, there are build systems that are commonly used in FTC, and none of those support mounting 1/8" square shaft directly. 

A servomotor is a type of motor, specifically one with a sensor of some sort on the output that is used to control it. By the common definition of servomotor, a NeveRest is more of a servo than a VEX 393. On the other hand, an FTC *Servo* is pretty much any motor that takes a 3-wire PWM input and does either positional or continuous control based on that. The VEX 393 does not fall under that general definition, and has to be explicitly defined as a *Servo*, but only legal when using a motor controller that does take a 3-wire PWM input. 

If you have specific recommendations for language that should be changed, feel free to open a pull request.

AlecHub, 2022-06-08:
So Game Manual 0 is the arbiter of what's competitive and what's not competitive.  I'm perfectly fine with that.  You are free to create any mission for yourself.

PR #239 (2022-06-07): Edit 393 and spm description
Closes #230