gamemanual0/gm0 #272: History of FTC and control systems

JSim011235, 2022-08-12:
**What information should be added?**Add a brief overview of important changes in FTC like control systems and major rules as well as unique moments and robots. This would be more of how FTC has changed and grown and why it is the way it is.

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
Appendix 
https://gm0.org/en/latest/docs/appendix/index.html

Eeshwar-Krishnan, 2022-08-13:
I think this is out of scope of gm0, however mentioning some of the history of the control system may be relevant to why certain things about the modern control system are what they are

Eeshwar-Krishnan, 2022-08-14:
Closing since this is out of scope, however this would be a cool idea for ftchistory (http://ftchistory.dogbuilt.net/) @abidingabi

closed 2022-08-14 by Eeshwar-Krishnan