gamemanual0/gm0 #363: I want to perform a translation

Teityan, 2023-06-12:
Sorry if I am posting here incorrectly.
I am looking to translate this manual into Japanese.
How should I perform the translation?
The easiest way is to fork it and replace it with Japanese, but that makes it harder to track when changes are made.

abidingabi, 2023-06-12:
gm0 has a Romanian translation, available at https://gm0.org/ro/latest/ through Transifex; see https://explore.transifex.com/gamemanual0/gm0/ and https://github.com/gamemanual0/gm0-translations/. It would be possible to request Japanese as a language on Transifex, and add it to gm0-translations, but I am not sure there is the demand for a complete Japanese translation.

Teityan, 2023-06-12:
I am a mentor doing a new FTC team in Japan. I have requested additional languages because I would like to translate for the future development of FTC in Japan and with my team members. I also want to translate gm0 because much of it is very good in FRC and other robotics competitions.