gamemanual0/gm0 #389: Add a translation version of Chinese Translations

DomnickJYH, 2023-09-09:
I’ve noticed the closed issue #363 asking for adding support for Japanese Translation, and I was inspired to create a Chinese translated version for gm0 then. I’m now the leader of my school’s robot club, my teammates and I are now working on the project of Chinese edition of gm0. We want to request for adding Chinese translations to gm0 because we would like to translate for the future development of FTC in China, for my club members and for some other FTC teams cooperating with us. I hope you may open the Chinese section soon and we can upload our Chinese translations.

DomnickJYH, 2023-09-09:
I’m not sure whether I am posting here in the right issue section. And I would be so excited if you guys allow us to upload a Chinese version of gm0, which may help Chinese FTC teams to read & understand gm0. And how could I upload the translation?

abidingabi, 2023-09-09:
Game Manual 0 uses transifex for translations. I have created a (currently empty) zh_CN translation; [you can request to join as a translator](https://app.transifex.com/gamemanual0/gm0/). There are some automation tools, but only translations that a human selects “review” on will be published at https://gm0.org/zh_CN/latest ~once a day.

DomnickJYH, 2023-09-10:
> Game Manual 0 uses transifex for translations. I have created a (currently empty) zh_CN translation; [you can request to join as a translator](https://app.transifex.com/gamemanual0/gm0/). There are [...]

Hello euhh, I've requested to join in the translation project and it seems that you need to permit my application. It says,

> 'Welcome DomnickJYH!
> With Transifex, you have everything you need to localize digital content and provide
> global customers with a beautiful user experience!
> 1 join requests have been sent to your projects of choice.
> In the meantime, [explore](https://explore.transifex.com/) other projects you can contribute to.'

I think you may have to check it, and then I'll be able to upload the translations.

abidingabi, 2023-09-10:
I have approved your request, and set you as a coordinator so that you can invite other people.

closed 2023-09-10 by abidingabi