gamemanual0/gm0 #416: Basic IMU Example

Krabbenhoft, 2023-12-06:
**What information should be added?**
I started working on a draft of possible IMU information. I am hoping to get feedback on if this is headed in the right direction and what major changes need to be made (I still need to review the style guide and maybe expand some things).

https://docs.google.com/document/d/1Pbkl8GGNRPcUUfDdw1RvUQzDddL5vIVZ5e9TMxxhzw8/edit

**Does it need it's own page? (Yes/No):** Yes

**If it needs a page, which section should it be in? If it doesn't, which page should it be under?**
[It should go under the programming tutorials section.](https://gm0.org/en/latest/docs/software/tutorials/index.html)

abidingabi, 2023-12-06:
Overall, it looks pretty good, but a couple preliminary notes:
- The turning code should probably be using a P controller
- It should also deal with turning in the shortest direction and what happens when you hit the point where the angles range

abidingabi, 2023-12-07:
> and include code that scales the speed according to the percentage of the turn completed?
This is pretty similar to what a P controller is :). A P controller sets the speed to `(target - setpoint) * kP`, where kP is a tuned constant, so that you slow down as you approach the target so you do not overshoot. Mentioning full-on PID control  would be smart, but I think a P controller is sufficient for this example.

Krabbenhoft, 2023-12-15:
I have worked more on it. What changes would you still like to see?

abidingabi, 2023-12-25:
Most of my suggestions would involve relatively small wording changes. It would be easier to deal with those if you opened a pull request for this, as I could use GitHub review functionality.

Krabbenhoft, 2024-01-05:
https://github.com/gamemanual0/gm0/pull/421